chandrawi/LoRaRF-Python #1: ImportError

BasMiienis, 2021-04-09:
Hello

I have installed the package on my rasberry pi to start using the waveshare SX1262_915M_LoRa_HAT 

however i get an import error
ImportError: cannot import name 'LoRaIO' from '__main__' (SX126x.py)

have you expierence this?

many thanks

regards

Bas

BasMiienis, 2021-04-09:
I am trying to use the SX126x_simple_node.py example and get the following error:

Begin LoRa radio
Traceback (most recent call last):
  File "lora.py", line 17, in <module>
    if not LoRa.begin() :
  File "/usr/local/lib/python3.7/dist-packages/LoRaRF/SX126x.py", line 351, in begin
    self.setSpi(self._bus, self._cs)
  File "/usr/local/lib/python3.7/dist-packages/LoRaRF/SX126x.py", line 331, in setSpi
    spi.open(bus, cs)
FileNotFoundError: [Errno 2] No such file or directory

chandrawi, 2021-04-13:
It most likely because SPI configuration on your raspberry pi is not correct. In SX126x_simple_node.py SPI bus id is 1 and cs is 0, but bus id 1 is not activated by default on newly installed raspberry pi OS.
To activate SPI with bus id 1 on raspberry pi you should edit /boot/config.txt

```
dtparam=spi=on
dtoverlay=spi1-3cs
```
After edit the file restart raspberry pi and check if SPI1 is already active using command
```
ls -l /dev/spidev*
```
If everything ok, then you will see
```
/dev/spidev0.0  /dev/spidev0.1  /dev/spidev1.0  /dev/spidev1.1  /dev/spidev1.2
```
Alternatively you can change your code to use SPI bus id to 0

chandrawi, 2021-04-13:
I found waveshare SX1262_915M_LoRa_HAT is using UART to connect to raspberry pi. So this package will not works. This package only works for connecting SX126x LoRa module using SPI.

closed 2022-01-20 by chandrawi