chandrawi/LoRaRF-Python #4: Migration from RPi.GPIO to gpiod

mgalemin, 2022-09-07:
Hi,

I'm looking for a python library for working with the SX1262 LoRa chip and I've found your library. It looks really good, but unfortunately, my product is based on the i.MX 6ULL processor and I can't use **RPi.GPIO** library in my project. I wonder how much effort would be required for me to migrate your library from **RPi.GPIO** to **gpiod** python library? I can see some dependencies on _remove_event_detect()_ and _add_event_detect()_ **RPi.GPIO** calls and I wonder how these calls can be implemented using **gpiod** calls? Or maybe you can recommend any other python libraries or wrappers that I could use for porting your library to a generic Yocto embedded linux build? Thanks.

Best regards,
Maks.

chandrawi, 2022-09-07:
Hi Maks,

Yes, add_event_detect() and remove_event_detect() can be implemented using gpiod. 
Actually, I planned to migrate from using RPi.GPIO to libgpiod library for the next version release, so we can use this LoRa library with Armbian and other linux based SBC. It is long way to go, but surely it will. There is python bindings in libgpiod we can use to replace RPi GPIO.
[https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/bindings/python](url)
Another way to use libgpiod is using pure python gpiod library.
[https://pypi.org/project/gpiod/](url)
I still figuring out which one we should use. What do you think Maks?

Best regards,
Chandra

mgalemin, 2022-09-08:
Hi Chandra,

I think porting your library to **gpiod** using pure python libraries is a much cleaner approach. The event detection with a callback can be implemented using **asyncio** library by adding a reader with your callback to an **asyncio** event loop _loop.add_reader()_ and feeding it with a file description of the **gpiod** _line_ class. The event FD can be retrieved by calling the _line.event_get_fd()_ method. This way you'll have only two dependencies on **gpiod** and **asyncio** python libraries.
I'll implement some PoC in a few days, maybe next week, as I need this functionality pretty urgently in order to de-risk my LoRa assumptions for the next project and I'll send you my version for review.

Best regards,
Maks.

chandrawi, 2022-09-09:
Ok maks, I am waiting for your version.

This is what Rpi.GPIO is used in the library:

1. Performing device reset by pulling NRESET pin to LOW.
2. Checking BUSY pin state before every SPI transfer.
3. Set TXEN to HIGH and RXEN to LOW when transmit and vice versa when receive. Also set TXEN and RXEN pin to LOW before sleep.
4. Set NSS pin to LOW to wake LoRa device in sleep mode.
5. For transmit and receive operation using interrupt, monitor RISING interrupt on DIO1/DIO2/DIO3 pin after putting LoRa device in receive mode and transmit mode to indicate transmit or receive process done.

I think it will be useful for you to porting the LoRaRF-python library.

mgalemin, 2022-09-12:
Hi Chandara,

I've implemented a proof-of-concept version of your library with gpiod library, please find it in the attachment. I've removed all the additional lines like the txen/rxen/wake as these are non-standard and are not present on the SX1262 chip. Also, all the LoRaRF methods shall be called from an asyncio task, so making it a generic library w/o asyncio dependency is not a straightforward task. But again, this was a quick fix just for my proof-of-concept LoRa integration for one of our products. Hope this helps.

Cheers,
Maks.

[LoRaPoC.zip](https://github.com/chandrawi/LoRaRF-Python/files/9545116/LoRaPoC.zip)