chandrawi/LoRaRF-Python #9: Question about hardware compatibility

langestefan, 2023-02-06:
Hi,

I have this board: [https://www.waveshare.com/sx1262-868m-lora-hat.htm](https://www.waveshare.com/sx1262-868m-lora-hat.htm)

Do you think it could work with this library? It seems to only support a UART based interface. On my module it says the type number is EBYTE E22 900T22S 1B. I have tried waveshare's supplied python code but it is very bad, their documentation is even worse. I can't even find a manual for this thing. I found a post on stackoverflow that says this board is not even using LoRa because it has a parameter called 'airspeed' which is not used in LoRa: [https://stackoverflow.com/questions/73852442/lora-communication-between-rpi-and-pico-with-waveshare-hats](https://stackoverflow.com/questions/73852442/lora-communication-between-rpi-and-pico-with-waveshare-hats). 

Are all EBYTE modules this bad or is mine just special? Do you think I could somehow make it work with LoRa? I have tried the supplied python code and put it into receiver mode but it doesn't receive any messages from my LoRa transmitters (STM32WL's)

chandrawi, 2023-02-06:
No, this library does not support LoRa module with UART interface like ebyte you have. I have tested some ebyte modules with SPI interface and they work fine.

Any LoRa module interfacing with SPI should works with this library. I suggest you use [this ebyte module](https://www.ebyte.com/en/product-view-news.html?id=436).

langestefan, 2023-02-06:
Hello, thanks for your reply. Do you think my module is a fake? If I look at the SX1262 datasheet it doesn't even have a UART interface, only SPI. This seems very fishy

chandrawi, 2023-02-07:
No, it is not a fake LoRa module. I think the module has a microcontroller so it can interface from LoRa chip to UART. The idea behind the UART LoRa or LoRaWAN modules are similar to a GSM modem which has UART interface. Like a GSM modem, an UART LoRa module has some commands to control module operation which is similar to AT command.