chandrawi/LoRaRF-Python #10: spi.open FileNotDoundError

mgh2010, 2023-02-18:
hi, I installed latest version of LoRaRF with pip on raspberry pi Zero 2 w, and did enabled the SPI.
but keep getting this error. what is Wrong.

[image]

chandrawi, 2023-02-23:
If you sure you have enabled SPI, `FileNotFound` error happen because the problem in Linux kernel, or you used wrong SPI bus you already enabled. Please note that if SPI is enabled and no issue in Linux kernel, this error is not going to happen even the SPI pin does not connect to the LoRa module. So, it is not caused by wrong wiring connection.

I guess the error happen because you enable SPI bus 0 and you did not change the SPI bus id in the example from 1 to 0. Also make sure you connect CS pin correctly.

chandrawi, 2023-05-14:
This error happen because the SPI port is not enabled yet for selected SPI
bus. If you used SPI port 1, add this line in /boot/config.tx

   1. dtoverlay=spi1-3cs