chandrawi/LoRaRF-Python #11: Receiving but not transmitting on 433.5 Mhz

VovaN60, 2023-02-26:
My system works with E220-M400S22 and E220-MM400S22 on various controllers like STM32F7, ESP32 with LoRaRF Arduino library.
I am able to receive and transmitt.
Trying to make it work with RPi with no luck...
I am able to receive messages from my other devices and see the success receive in terminal of RPi.
As well as i see that RPi sends the messages.
`Hi from Base Station!  125`
`Packet status: RSSI = -36.00 dBm | SNR = 13.25 dB`
`HeLoRa from Raspberry Pi!  125`
`Transmit time: 19.69 ms | Data rate: 1371.37 byte/s Int status: 1`
But other side devices doesn't receive them.
I can confirm the receiver gets messages from other devices at the same time (I use total 3 devices for this test included RPi).

Here is my script:
[image]

chandrawi, 2023-02-26:
So the problem is LoRa package transmitted by raspberry pi does not received by other devices while raspberry pi can received from other devices?
In this case you should check `LoRa.setModulation`, `LoRa.syncWord`, and `LoRa.setFrequency` have same parameters on the transmitter and receiver. You also should check your code on the receiver, if there any chance while the package transmitting the receiver doing something else than receiving the package.

VovaN60, 2023-02-27:
Dear @chandrawi ,

Yes i have checked and rechecked many times already.
I even compared the libs how they operate registers write and read when doing this functions.
All same...
I am sure my Base Station is listening when the RPi sending packet. Any time i send message from other device (3d device on ESP32) the base station receives the message. But when i send from RPi - not receiving...
There is LLCC68 in my modules. Should be same as SX126x.
Kindly inform if you successefully used this library with 433 modules.

chandrawi, 2023-02-27:
Dear @VovaN60,

I have successfully used this library using SX1262 lora module with 433 Mhz frequency. The test include transmitter on raspberry pi and receiver on Arduino.

Are all the receivers using LLCC68? I never used the LLCC68. But someone has tested it using the LoRaRF-Arduino library and works. Unfortunately, no one has reported the transmission between Raspberry pi and Arduino using LLCC68 is working.

Another thing you should check is the 3.3V power supply you used. Please note that transmit process is more consumed power than receive. 3.3V from Raspberry pi GPIO is not recommended for LoRa module. Use external 3.3V or use 5V from GPIO with 3.3V linear regulator rated >= 500 mA.

VovaN60, 2023-02-27:
Dear @chandrawi 

Yes all the receivers and transmitters are LLCC68. 
I use E220-M400S22 and E220-MM400S22 EByte modules.

Yes i work with LoRaRF-Arduino library with this modules on various controllers. All works.

Thank you very much for advice.
Currently the module is powered from RPi 3.3v pin.
I will test with external power supply today and inform.

VovaN60, 2023-02-27:
Dear @chandrawi 

I confirm the problem was in power supply. Thank you for an advice.
But not the 3.3v header.
The GND was weak....

So now the module is working as expected with power connected to normal 3.3v header on RPi.

Thank you one more time.

Mr-Essid, 2024-04-26:
hi @VovaN60 I have the same issue can you share with us how did you fix it please,

VovaN60, 2024-04-28:
> hi @VovaN60 I have the same issue can you share with us how did you fix it please,

Hi.

Just make power connection to the module strong.
Take GND from strong GND on your board. If PCB already designed - cut the existing GND connection and solder the external wire from abttery or power source directly to the module. Should help.
If GND doesn't help - do the same with 3.3v source.

Best regards,
Vladimir.