chandrawi/LoRaRF-Python #12: The library doesn't work with multipple modules.

VovaN60, 2023-05-02:
Hello Dear,

The lib doesn't support multipple SPI LoRa modules.
When i create a multipple SX126x() objects all seams to work. But in reality it works with only 1 module.
Here what i modified in the library and now it is all works:
Removed this:
`spi = spidev.SpiDev()`
Added this:
`self._spi = spidev.SpiDev()` 
to def setSpi function of the object.

Please confirm it is right. And may be add to the lib. So people can use multiple modules.

chandrawi, 2023-05-15:
Hello dear, 

Actually you can use multiple SX126x object to transmit and receive. 

You just need call `setActive()` method on the object before call `beginTransmission()` or `request()` method.

VovaN60, 2023-05-15:
Dear @chandrawi 

I think you are talking about Arduino SPI work.
I am running your library on RPi.