chandrawi/LoRaRF-Python #16: Receiver not working

Jorge-E-Guerrero, 2023-07-25:
Hi,

I'm trying to run the proyect on a Raspberry Pi 4 with Raspberry OS 32 bits with the Semtech SX1262 LoRa Module. 

When running the receiver_listen.py or receiver_continuous.py shows the following message:

Begin LoRa radio
Set RF module to use TCXO as clock reference
Set frequency to 915 Mhz
Set RX gain to boosted gain
Set modulation parameters:
        Spreading factor = 7
        Bandwidth = 125 kHz
        Coding rate = 4/5
Set packet parameters:
        Explicit header type
        Preamble length = 12
        Payload Length = 15
        CRC on
Set syncronize word to 0x3444

-- LoRa Receiver Listen --

#]©(nBcâ-¨ëüó?¨V´ã£�xº9ûß▒Ï{  98
Packet status: RSSI = -29.50 dBm | SNR = 10.50 dB

It seems like the LoRa Radio works fine, but is not receiving any real message, and shows an initial corrupt message. Im sending packets from an arduino program with the same sync word, frecuency and bandwidth.

I've tried using both SPI bus but the results are the same.

chandrawi, 2023-07-26:
Is the corrupt message only happen once and then correct message received? Or the corrupt message always happen? Do you use Explicit or Implicit package type?

Jorge-E-Guerrero, 2023-07-26:
It happens everytime I run the script and just receive that corrupt message. I use the explicit package. Also I tried to run the transmitter.py but the receivers still not working.

aditya-arv13, 2023-08-11:
I am also experiencing the same problem with receiver.py. The corrupt message is received once and then it struck forever with no messages.

NistantriTech-Dhaval, 2024-08-14:
> Hi,
>
> I'm trying to run the proyect on a Raspberry Pi 4 with Raspberry OS 32 bits with the Semtech SX1262 LoRa Module.
>
> When running the receiver_listen.py or receiver_continuous.py shows the [...]

Please share your pin connection

johannesvwyk, 2024-09-11:
I've been trying to make this work for more than 3 days now. I get the lora ra01-H to send a message or well the code shows a time stamp and the example codes bit rate and stuff. I cant seem to get the receiver side to get the message. I have not changed any of the code so I am getting desperate for some help. I'm running it off a raspberry pi zero and a 3b+. the receiver also starts without errors but there is no incoming messages, I hope this reaches someone that is willing to assist me in making this work as it would mean a lot to me! 

further information:
i'm using the SX127x example code and my pin layouts for both are
GND = 6
3.3V = 1
NSS = 24
MOSI = 19
MISO = 21
SCK = 23
RESET = 15
DI01 = 16

BNNorman, 2024-09-12:
Just a thought.

How close are your transmitter and receiver? Too close and the receiver inputs will be swamped such that receiver decoder circuits will just see jibberish.

I have used RadioHead server/ client successfully with about 3m between using rfm95. A brick wall between tx and rx would help a lot.

johannesvwyk, 2024-09-13:
Hi, I appreciate the response! I managed to get my coding done up to phase one. It turned out to be my frequency! I was not aware that each region had its own set frequency that you have to use. The sample said 915e6 and I had to use 868e6. I might be wrong but somehow it worked! Next up is either using a raspberry pi pico as the transmitter or skipping that step and using a ATtiny.