chandrawi/LoRaRF-Python #17: LoraRF - SX126X - set device EUI

bejanvaleriu, 2023-10-06:
Hi,

I'm currently working on a project that involves sending data from SX126x (connected to raspberry pi 4b) to a Mikrotik Lora Gateway and then to ChirpStack. Browsing through multiple docs and projects I found that I need a Device EUI set on my SX126x in order to connect it to the server in order to receive data from it. On this repo I did not find any referece to Device EUI. Can you please help me with my project?

Regards,

MaffooClock, 2023-12-11:
I am also attempting this exact thing -- a MiroTik R11e-LR9 that is connected to ChirpStack, but also The Things Stack.

Looks like [examples/network](https://github.com/chandrawi/LoRaRF-Python/tree/main/examples/network) is close to what we're looking for: it uses a [struct](https://docs.python.org/3/library/struct.html) to assemble data fields into a binary packet.  But it's not exactly LoRaWAN, since it doesn't include OTAA or ABP functions (which require an App ID and Key, along with the Device EUI as you pointed out).

I think what we're going to have to do is learn how to implement OTAA or ABP ourselves so that we can join the MikroTik gateway, and then learn how to construct a raw LoRaWAN packet.

In other words, LoRaRF doesn't directly support LoRaWAN, but I think we can implement LoRaWAN ourselves with the help of some of this library's functions.

I've used LoRaWAN on Arduino, so my approach to this will be to see how [one of those libraries](https://github.com/arduino-libraries/MKRWAN/blob/master/src/MKRWAN.h) work so I can replicate that in Python with this library.

pestu1, 2024-08-05:
Did you get this working? I'm trying to join a chirpstack gateway but can't find a propper python example.

Mat3814, 2025-07-21:
Does someone found a way to support LoRaWAN with this driver ? I am currently working on it but not sure if it's possible