chandrawi/LoRaRF-Python #18: Network Examples working

pramsuiitd, 2023-10-15:
Hi, I am using LoRa SX1262 Raspberry Hat with Raspberry Pi. This is the issue I am getting when using the Network Examples reciever.py code

Begin LoRa radio
Traceback (most recent call last):
  File "tx.py", line 15, in <module>
    if not LoRa.begin() :
  File "/usr/local/lib/python3.7/dist-packages/LoRaRF/SX126x.py", line 293, in begin
    self.reset()
  File "/usr/local/lib/python3.7/dist-packages/LoRaRF/SX126x.py", line 310, in reset
    self._reset.output(LoRaGpio.LOW)
  File "/usr/local/lib/python3.7/dist-packages/LoRaRF/base.py", line 36, in output
    chip = gpiod.Chip(self.chip)
AttributeError: module 'gpiod' has no attribute 'Chip'

I believe that the module gpiod has 'chip' attribute instead of 'Chip' attribute. Can you verify and fix this in a patch?

gostower, 2023-11-02:
Hi, I'm having the same issue, have you find any solution ?

Thanks

pramsuiitd, 2023-11-04:
> Hi, I'm having the same issue, have you find any solution ?
> 
> Thanks

Hi @gostowerx, 
I have not found a solution. I tried to change the 'Chip' with 'chip' everywhere in the code, but it led to other errors. Please let me know if you find something.
@chandrawi can you please have a look at the issue and patch?
Thanks

MaffooClock, 2023-12-08:
I'm guessing you installed the [python3-gpiod](https://github.com/hhk7734/python3-gpiod) module.  What you actually need is the [libgpiod](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/about/) module.

If you're on Ubuntu/Debian, just run `sudo apt install python3-libgpiod`.