chandrawi/LoRaRF-Python #19: Error: cannot import name 'LoRaSpi' from 'LoRaRF'

vinityadava, 2023-12-07:
I was installing the network node from network folder  , but it throws this error when pressed run. Can anyone resolve it?

MaffooClock, 2023-12-08:
You must be following the current [examples](https://github.com/chandrawi/LoRaRF-Python/tree/main/examples/network); but those were refactored (commit 67cb5c9) after the v1.4.0 release (commit 86572c3), so they reference a new API that doesn't exist in the published version.

You have two options:

1. Install the bleeding-edge version with:
   ```shell
   pip3 install git+https://github.com/chandrawi/LoRaRF-Python.git
   ```
   Note that you might get a `No module named 'gpiod'` exception with this version, so you'll need to be prepared to deal with that (installing [python3-gpiod](https://github.com/hhk7734/python3-gpiod) doesn't seem to be the right way, as I get the same `module 'gpiod' has no attribute 'Chip'` errors as in Issue #18).

2. Stay with the current v1.4.0, but reference [the examples from that version](https://github.com/chandrawi/LoRaRF-Python/tree/86572c34f90435b24d71f8fd364ff0043fd7b51e/examples).

vinityadava, 2023-12-08:
Thank you for that, I followed the same.

but now it says,

[image]

MaffooClock, 2023-12-08:
Do you not have SPI enabled?

If you run `ls /dev/spi*`, you should see something like this:
[image]

...if not, then you don't have SPI enabled (and that's why you see a "file not found" error).

MaffooClock, 2023-12-08:
> I am using raspberrypi Zero w with Lora module SX1262

Same!  And mine is working.

After looking closer at the error message, I see a reference to `gpiod/chip.py`, but you shouldn't be using a module called "gpiod," I don't think -- I don't have that in my environment, anyway.  Seems like your Python environment might be goofy.  But I'm just guessing, not sure what else to advise.

MaffooClock, 2023-12-11:
I'm using DietPi, 64-bit (bookworm).  But its Apt uses the Debian repos, so it'd be the same as Raspbian (or Raspberry Pi OS).  But I don't think this is an OS issue.

I just noticed that LoRaRF is using `gpiod`:
https://github.com/chandrawi/LoRaRF-Python/blob/67cb5c96fd4bb95db57e176590ced4b78b3f1192/LoRaRF/base.py#L2
... so I was half-way wrong earlier.

So what happens if you run this?
```bash
python3 -c "import gpiod; print( gpiod.__file__ )"
```

For me, it returns `/usr/lib/python3/dist-packages/gpiod.cpython-311-aarch64-linux-gnu.so` -- a compiled shared object, not a straight Python file (which is consistent with [what the Debian package installs](https://packages.debian.org/bookworm/arm64/python3-libgpiod/filelist) for my system).

I would expect yours to return `/usr/lib/python3/dist-packages/gpiod.cpython-311-i386-linux-gnu.so`, and if so, then maybe just delete `~/.local/lib/python3.9/site-packages/gpiod`?

vinityadava, 2023-12-13:
Thanks @MaffooClock I resolved the issue by following above steps, and succesfully transmitting and receiving data.

Now i want to send the list of floats but it is showing the error Type error.
I have mentioned it in issue 13 /issue #21 

Can you help about it?

vrishabkakade, 2024-07-15:
> I'm using DietPi, 64-bit (bookworm). But its Apt uses the Debian repos, so it'd be the same as Raspbian (or legacy Raspberry Pi OS). But I don't think this is an OS issue.
> 
> I just noticed that LoRaRF is using `gpiod`:
> 
> https://github.com/chandrawi/LoRaRF-Python/blob/67cb5c96fd4bb95db57e176590ced4b78b3f1192/LoRaRF/base.py#L2
> 
> 
> ... so I was half-way wrong earlier.
> So what happens if you run this?
> 
> ```shell
> python3 -c "import gpiod; print( gpiod.__file__ )"
> ```
> 
> For me, it returns `/usr/lib/python3/dist-packages/gpiod.cpython-311-aarch64-linux-gnu.so` -- a compiled shared object, not a straight Python file (which is consistent with [what the Debian package installs](https://packages.debian.org/bookworm/arm64/python3-libgpiod/filelist) for my system).
> 
> I would expect yours to return `/usr/lib/python3/dist-packages/gpiod.cpython-311-i386-linux-gnu.so`, and if so, then maybe just delete `~/.local/lib/python3.9/site-packages/gpiod`?

MaffooClock,
Thank you for this. It helped me troubleshoot my issue as well. But I am running this in a virtual environment and deleting the gpiod directory from the virtual environment simply throws the below error

`(weewx-venv) weewx@freedompi:~ $ python3 -c "import gpiod; print( gpiod.__file__ )"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'gpiod'
`
Contents of the gpiod directory
`(weewx-venv) weewx@freedompi:~/weewx-venv/lib/python3.11/site-packages/gpiod $ ls -lrat
total 324
-rw-r--r--  1 weewx weewx    195 Jul 10 13:22 version.py
-rw-r--r--  1 weewx weewx   2222 Jul 10 13:22 line_settings.py
-rw-r--r--  1 weewx weewx   6911 Jul 10 13:22 line_request.py
-rw-r--r--  1 weewx weewx   1262 Jul 10 13:22 line.py
-rw-r--r--  1 weewx weewx   2259 Jul 10 13:22 line_info.py
-rw-r--r--  1 weewx weewx    503 Jul 10 13:22 internal.py
-rw-r--r--  1 weewx weewx   1419 Jul 10 13:22 __init__.py
-rw-r--r--  1 weewx weewx   1125 Jul 10 13:22 info_event.py
-rwxr-xr-x  1 weewx weewx 256256 Jul 10 13:22 _ext.cpython-311-aarch64-linux-gnu.so
-rw-r--r--  1 weewx weewx    545 Jul 10 13:22 exception.py
-rw-r--r--  1 weewx weewx   1371 Jul 10 13:22 edge_event.py
-rw-r--r--  1 weewx weewx  10956 Jul 10 13:22 chip.py
-rw-r--r--  1 weewx weewx    476 Jul 10 13:22 chip_info.py
drwxr-xr-x  2 weewx weewx   4096 Jul 10 13:22 __pycache__
drwxr-xr-x  3 weewx weewx   4096 Jul 15 12:40 .
drwxr-xr-x 52 weewx weewx   4096 Jul 15 12:48 ..
`

How do I get my program to pick up /usr/lib/python3/dist-packages/gpiod.cpython-311-aarch64-linux-gnu.so when I'm working in a virtual environment? I have tried creating a link, but that doesn't work either.