chandrawi/LoRaRF-Python #20: Dependency and pip version problem

rainerfritz, 2023-12-10:
Hi!

I try to get the library running on an RPi 3b+ with the 64bit OS headless. The board I am using is the waveshare board with SX1262 via SPI and GPS via serial. 

When I compare the version if install directly LoRa RF via pip3 install LoRaRF, the base.py file seems not uptodate as it misses the classes LoraSPI and LoraGpio. So i am going with the version here from github, manual install in an venv.

I do the following:
-create virtual env for python
`python -m venv lora`

--activate if not already
`source lora/bin/activate`

--get the LoRa Repo from git - pip is not actual
```
cd lora
git clone https://github.com/chandrawi/LoRaRF-Python.git
```

--install dependencies
```
pip3 install wheel
pip3 install spidev
pip3 install gpiod
```

--install LoRaRF
```
cd LoRaRF-Python
python3 setup.py bdist_wheel
pip3 install dist/LoRaRF-1.4.0-py3-none-any.whl
```

--try to call the receiver example:
`python3 examples/SX126x/receiver.py`

Gives the Error:
```
Begin LoRa radio
Traceback (most recent call last):
  File "/home/rainer/lora/LoRaRF-Python/examples/SX126x/receiver.py", line 15, in <module>
    if not LoRa.begin() :
           ^^^^^^^^^^^^
  File "/home/rainer/lora/lib/python3.11/site-packages/LoRaRF/SX126x.py", line 293, in begin
    self.reset()
  File "/home/rainer/lora/lib/python3.11/site-packages/LoRaRF/SX126x.py", line 310, in reset
    self._reset.output(LoRaGpio.LOW)
  File "/home/rainer/lora/lib/python3.11/site-packages/LoRaRF/base.py", line 36, in output
    chip = gpiod.Chip(self.chip)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/home/rainer/lora/lib/python3.11/site-packages/gpiod/chip.py", line 58, in __init__
    self._chip = _ext.Chip(path)
```
                 ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory

Can it be, that the actual gpiod version 2.1.3 does not work with the code currently?

I changed then in the chip.py the line `self._chip = _ext.Chip(path) `to `self._chip = _ext.Chip("/dev/gpiochip0") `which seems to work, but then the chip object complains:

```
Begin LoRa radio
Traceback (most recent call last):
  File "/home/rainer/lora/LoRaRF-Python/examples/SX126x/receiver.py", line 15, in <module>
    if not LoRa.begin() :
           ^^^^^^^^^^^^
  File "/home/rainer/lora/lib/python3.11/site-packages/LoRaRF/SX126x.py", line 293, in begin
    self.reset()
  File "/home/rainer/lora/lib/python3.11/site-packages/LoRaRF/SX126x.py", line 310, in reset
    self._reset.output(LoRaGpio.LOW)
  File "/home/rainer/lora/lib/python3.11/site-packages/LoRaRF/base.py", line 38, in output
    line = chip.get_line(self.offset)
           ^^^^^^^^^^^^^
AttributeError: 'Chip' object has no attribute 'get_line'. Did you mean: 'get_info'?
```

Any help to get it running would be awesome.
THX, Rainer

JulianGa05, 2024-01-13:
Hello Rainer,

I am currently facing the same issue. Have you managed to fix it yet?

Best regards
Julian

rainerfritz, 2024-01-14:
I try to get the waveshare lora hat running with SX1262 and gps. Took me a little bit to find the right documentation. They provide a LoRaRF version, which works out of the box. I had no time so far to have a look where they modified it. If you want to have a look in the meanwhile here is the link. Scroll down to the bottom of the page: [waveshare lorarf library](https://www.waveshare.com/wiki/SX1262_XXXM_LoRaWAN/GNSS_HAT)

JulianGa05, 2024-01-16:
Wow, how could I have not found that earlier? Thank you for that! I was
always just relayed to a different product page that had only useless demo
code.
To be honest though, I am still not able to make it work. I've tried to use
the transmitter.py example and at first I got an error with the call to
SX126x() and after I fixed that I got an error during the LoRa.begin().
Have you encountered similar issues?

JulianGa05, 2024-01-16:
> Before you test it with that library, make sure you uninstall from your
> primary python environment the lorarf package.
> Thats how I did it:
>
>    - generate a virtual python environment
>    - activate it and cd into it
>    - copy the whole library into the venv directory
>    - pip install ./sx126x_lorawan_hat_code/python/lora
>    But it worked also without installing it on my side

That was it, I still had the old installation, thank you so much!! Now it
works!

longtomin, 2024-09-12:
Hello,

I got the same module for my Model: Raspberry Pi 4 Model B Rev 1.1.
When I try it, with a venv, I can not install the lib from the zip file of waveshare, e.g. `python3 setup.py install` in the directory sx126x_lorawan_hat_code/python/lora gives me error like this:
`Processing spidev-3.6.tar.gz
Writing /tmp/easy_install-tksjunjd/spidev-3.6/setup.cfg
Running spidev-3.6/setup.py -q bdist_egg --dist-dir /tmp/easy_install-tksjunjd/spidev-3.6/egg-dist-tmp-ticyfavm
/home/thomas/lora/lib/python3.11/site-packages/setuptools/dist.py:775: UserWarning: Usage of dash-separated 'description-file' will not be supported in future versions. Please use the underscore name 'description_file' instead
  warnings.warn(
/home/thomas/lora/lib/python3.11/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
Traceback (most recent call last):
  File "/home/thomas/lora/lib/python3.11/site-packages/setuptools/sandbox.py", line 156, in save_modules
    yield saved
  File "/home/thomas/lora/lib/python3.11/site-packages/setuptools/sandbox.py", line 198, in setup_context
    yield`

If I put a sudo in front of the python3, I get these error message:
`sudo python3 setup.py install
Traceback (most recent call last):
  File "/home/thomas/lora/sx126x_lorawan_hat_code/python/lora/setup.py", line 1, in <module>
    import setuptools
ModuleNotFoundError: No module named 'setuptools'`

The Lib from the ZIP File runs without installing it, but the target is to have a long running (multiple months/years) solution. If I run it the python process, it sometimes stopping without any failure. The data send over LoRa is no longer processed. Making the python script as a serivce was also not successful because of the lacking of the installation, which is preventing the python script to start.

At the end I cloned this git repository and set the pins to the numbers named in the example scripts of the ZIP file.
It will start, but I get strange error messages like Index out of range by returning the buf[0] in SX126x.py line 777 of the read function `if single : return buf[0]` using the modified (Pins) receiver_continous.py script.

I'm confused, thanks for you help.

longtomin, 2024-09-13:
Hello,

I tried the EByte Module for the RPi4 which has an AirRate and no spreading factor or Bandwidth.
So this was useless. Then I found this hat which is using this library.
What I want to build is a XXL weather station for several places (garden, garage, living room, bath room, .....)
So I got several Heltec Wireless Stick Lite Modules with LoRa on board, this works quit well even with batteries and a monitoring of the battery voltage via LoRa.
What is a mess is, I can not get the RPi4 in a stable mode which can do long running tasks.
I setup the RPi Image without a desptop (not needed).
I tried the installation via pip3 (message: use external vEnv)
Made a vEnv and tried the pip3 install (message: #include python.h missing)
clone the git from here and make a install with changed pins (Message: Tuple out of range)

Is there a step by step installation guide from a brand new installed RPi Image, e.g.

1. Activate SPI over raspi-config (sudo raspi-config) Interface options - SPI - Activate
2. Update the RPi Image sudo apt-get update; sudo apt-get upgrade -y; sudo apt-get dist-upgrade -y; sudo apt-get autoclean -y; sudo apt-get autoremove -y
3. Install Package xyz (needed for build or install)
4. Make a new vEnv where the Python stuff is going to...... or remove/rename the /usr/lib/python3.11/EXTERNALLY-MANAGED file to be able to install directly.
5. Run the example scripts to test with the pin setup, e.g. ......

Thanks & regards

JulianGa05, 2024-09-13:
Ah yes, I was talking about my experiences with the hat with the Ebyte module. Sorry, I should have made this clearer. I have no experiences with this hat but I wish you the best of luck in finding a solution!

longtomin, 2024-09-13:
If you make a diff on the SX126x.py from the ZIP file and from the repository here, there is not much difference, but never then less it's not working

rainerfritz, 2024-09-13:
Are you sure you have this zip downloaded?
[Sample Code](https://files.waveshare.com/wiki/SX1262-XXXM-LoRaWAN-GNSS-HAT/Sx126x_lorawan_hat_code.zip)

I did not install it with the setup.py, I copied the library in the into the venv and installed it then with pip with the install command:
`pip install ./sx126x_lorawan_hat_code/python/lora`

longtomin, 2024-09-13:
Yes, I downloaded the ZIP (sha256sum is 8273f7605d73a7353311282faa69d2334b54af3736d09144bef06a6801aeee14) on both machines, my Ubuntu Laptop and my RPi4.
It is running even without installing it, just unzip and put the Python script into the example/SX126x folder.
So it is running and receives data. The tricky point is now to process the data and do it 24/7. In my case saving the data into a local influxdb to be able to show them and make some calculations. But here it stops. Sometimes the python process crashes sometimes it stops writing data into the DB.
If I setup a local service (https://medium.com/codex/setup-a-python-script-as-a-service-through-systemctl-systemd-f0cc55a42267) it will not start because the influxdb client is missing and I had no chance to bring both (inluxdb-client and LoRaRF) together. Not with a vEnv nor with the renaming of the /usr/lib/python3.11/EXTERNALLY-MANAGED file.
I did also a vEnd and made the pip3 install LoRaRF, but in the base.py the LoRaSPI and LoRaGPIO is missing even in the version 1.4.0.

If you make a diff on the SX126x.py from the ZIP file and from the repository here, there is not much difference, but never then less it's not working.

longtomin, 2024-09-17:
Hello,

I found a solution for my Hat with the RadioLib compiled by cmake on the Raspberry with C and the functions of the Lib which are also used on the Arduino.
See: https://github.com/jgromes/RadioLib/tree/master/examples/NonArduino/Raspberry
And for the HAT I used these parameters:
// create a new instance of the HAL class
PiHal* hal = new PiHal(1);
// now we can create the radio module
SX1262 radio = new Module(hal, 21, 16, 18, 20);

rainerfritz, 2024-09-18:
Hi longtomin!

Saw this possibility also and tried it out myself but I can't get the lora chip initialised, getting error code -2. I tried it with both SPI channels 1 and 0 as the python version worked with channel 0,0.

My workflow was to compile radiolib and lgpio and linked in the Cmakeslists.txt to the header files. Did the following:

```
wget http://abyz.me.uk/lg/lg.zip
unzip lg.zip
cd lg
make
sudo make install
```

gets installed in: /usr/local/include

radiolib ziehen entweder zip via wget oder git clone

```
$ mkdir build/
$ cd build
$ cmake ..
$ sudo make install
```

gets installed in: /usr/local/include/RadioLib

`cd RadioLib-master/examples/NonArduino/Raspberry`

Edit CMakeLists.txt

`nano CMakeLists.txt`

comment out the add_subdirectory line because all header files are compiled as shared library

add:

#link to shared library build before
```
include_directories("/usr/local/include")
include_directories("/usr/local/include/RadioLib")
```

How did you get it working?

BR
Rainer

longtomin, 2024-09-18:
Hello,

I just installed the 2024-07-04-raspios-bookworm-arm64.img on a SD-Card or USB Stick and assigned a static IP with the eth0 file in /etc/network/interfaces.d direcotry.
Afterwards I made a `sudo raspi-config` and I activated the SPI with 3 - Interface Options -> 4 SPI (Enable).
The I update the Image with: `sudo apt-get update; sudo apt-get upgrade -y; sudo apt-get dist-upgrade -y; sudo apt-get autoclean -y; sudo apt-get autoremove -y`
Afterwards I just installed git with `sudo apt-get install git`
Then I clone the Repositoy with `git clone https://github.com/jgromes/RadioLib.git`
In the examples/NonArduino/Raspberry I just changed the line 34 of the main.cpp to my parameters, in my case: `SX1262 radio = new Module(hal, 21, 16, 18, 20);`

And I had to change the radio.begin like this:```

#define FREQUENCY 868.0
#define BANDWIDTH 125.0
#define SPREADING_FACTOR 7
#define TRANSMIT_POWER 0
#define CODING_RATE 5

int state = radio.begin(FREQUENCY, BANDWIDTH, SPREADING_FACTOR, CODING_RATE, 0x12, 10, 8, **0.0**, false);

Important is the 0.0 in the radio.begin.
```

rainerfritz, 2024-09-19:
Awesome! Thanks, the 0.0 and false at the end did the trick to not use TCXO and LDO.