chandrawi/LoRaRF-Python #24: No module named 'gpiod'

AsgerKennet, 2024-05-03:
Im trying to run the reciver.py in examples but is getting the error "no module name gpiod"
Its the same in ["this issue"](https://github.com/chandrawi/LoRaRF-Python/issues/19)..

I have tried to write this
`(LoRa) toor@kennet:~ $ python3 -c "import gpiod; print( gpiod.__file__ )"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'gpiod'
`

But have install python3-libgpiod

Excalibur888, 2024-05-07:
Try `pip install gpiod` in your venv
or `pip install gpiod --break-system-packages` if you are not in a virtual environement

AsgerKennet, 2024-05-07:
I have already tried so in another virtual environement but some classes and methods are missing..
But I copyed the `/usr/lib/python3/dist-packages/gpiod.cpython-311-aarch64-linux-gnu.so` from the `python3-libgpiod`
to my new environement and it works - kinda...
Somehow it will not recive my LoRa signal from my microcontroller