chandrawi/LoRaRF-Python #28: ModuleNotFoundError: No module named 'gpiod.line'; 'gpiod' is not a package

Praveenraj725, 2025-01-30:
i am trying to run the example sx1278 receiver.py but i get this error:

iot@raspberrypi:~/test2/LoRaRF-Python/examples/SX127x $ sudo python3 receiver.py
Traceback (most recent call last):
  File "/home/iot/test2/LoRaRF-Python/examples/SX127x/receiver.py", line 4, in <module>
    from LoRaRF import SX127x, LoRaSpi, LoRaGpio
  File "/home/iot/test2/LoRaRF-Python/LoRaRF/__init__.py", line 2, in <module>
    from .base import LoRaSpi, LoRaGpio
  File "/home/iot/test2/LoRaRF-Python/LoRaRF/base.py", line 3, in <module>
    from gpiod.line import Direction, Value, Edge
ModuleNotFoundError: No module named 'gpiod.line'; 'gpiod' is not a package

KimiKimar27, 2025-02-27:
Did you find a solution?

villain-98, 2025-03-16:
You have to run the following command:
`pip3 install gpiod`
In case you encounter problem installing this, simply run:
`pip3 install gpiod --break-system-packages
`

[https://pypi.org/project/gpiod/](url)